v002/v002-Camera-Live #96: EOS 500D - Camera Error

mattryyc, 2020-04-08:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: [10.15.3 (19D76)
 - Camera manufacturer and model [Canon EOS 500D / T1i]
 - Camera Live version 13

**Describe the bug**
When connecting the camera over USB, the camera is detected, but after connecting goes to 'Camera Error'

**To Reproduce**
0. Open Camera Live
1. Connect Camera over USB
2. Set camera wheel to video.
3. Turn on Camera.
4. Camera Live Shows Canon Eos 500D, but shows Starting briefly then Camera Error.
**Expected behavior**

1. Camera status shows 'Ready'

bangnoise, 2020-04-08:
There have been reports of success with the 500D - so it's probably a problem with the USB connection. See the [Troubleshooting section](https://github.com/v002/v002-Camera-Live#troubleshooting) of the README.

megerman, 2020-07-06:
I also have this problem and have tried multiple cables - I don't think it's a USB connection issue.

mattryyc, 2020-07-06:
My issue was resolved by moving to another mode (I use AV mode). Video mode on the EOS500D does not support Live View.

megerman, 2020-07-07:
I built locally and was able to fix the problem

xiaokunx, 2021-12-17:
> My issue was resolved by moving to another mode (I use AV mode). Video mode on the EOS500D does not support Live View.

That worked for me too! Thanks for the tip!