v002/v002-Camera-Live #101: Canon M50 produces error after 10-12 minutes

blowtorch991, 2020-04-12:
System:
macOS version: [10.15.4]
Camera manufacturer and model [Canon EOS M50]
Camera Live version 13

Describe the bug:
When connecting the camera over USB, the camera is detected and begins to run, but after approximately 10-12 minutes, Camera Live produces a 'Camera Error' and the image is frozen. Must turn off camera and restart to fix error. I am using battery adapter and direct power from wall outlet.

To Reproduce:
Open Camera Live
Connect Camera over USB
Set camera wheel to manual video.
Turn on Camera.
Camera Live shows M50 video capture, but after 10-12 minutes produces a Camera Error.

blowtorch991, 2020-04-13:
Also to note, the timing of this happens inconsistently. Sometimes it's 5 minutes, other times it's 10 minutes or longer.

blowtorch991, 2020-04-14:
I also tested with a functionally similar software called Ecamm Live and it works for 30+ minutes.

frankprogrammer, 2020-04-16:
I am getting the error on my M50 with both firmwares 1.0.1 and 1.0.3. Running macOS Catalina 10.15.2. Same 5-10 minute window. Got this with a battery and also with a battery adapter. 1080p, 30fps, 1/60 shutter.

bangnoise, 2020-04-19:
Are there any images on the camera's memory card? Does using a memory card with no images make a difference?

blowtorch991, 2020-04-19:
Just removed the memory card - testing now. Will report back asap after testing.

blowtorch991, 2020-04-19:
No memory card in the camera did not provide any significant changes to the camera error after 5+ minutes. Tried four separate times resulting in the same error. The live feed actually disappeared from the Camera Live program and from Quicktime. However, I only had to tap on the shutter button and it popped back up. Seems like it could be an inactivity problem/timing out problem.

blowtorch991, 2020-04-21:
I posted on a reddit feed and someone responded to my question by sharing that walking back to v10 fixes this issue. I've used it several times now and for longer periods 30-60+ minutes and have not run into any problems.

boggsey, 2020-05-12:
Thanks @blowtorch991 - Right now, your suggestion is working for me, as well.

I'm running an M50 with no SD card and powered through the wall. I ran about 10 tests with and without the card on v13 and it crashed inconsistently within 10-20 minutes each time. Rolled back to v10 per the above comment and have been able to stay connected for around 2 hours consistently.

Jip-Hop, 2020-05-13:
Tried version 13 (Alpha) with my Canon M50 (with SD card) + power adapter and 22mm F2 lens in combination with OBS and it stopped working for me after about 15 minutes. On the camera display it no longer showed the live view, but instead a black screen with PC connection icon. Additionally the lens focus motor was moving constantly... Then tried with version 10 and the 15-45mm kit lens and it's been running fine for 1 hour.

I'm on MacOS 10.14.6.

Jip-Hop, 2020-05-14:
On MacOS 10.15 with my Canon M50 and the 15-45mm kit lens I found version 11 to work for a long time.

ShKlinkenberg, 2020-08-12:
Was having the same problem. Getting canon err 70 after between 5 to 15 minutes. After extensive trouble shooting, i narrowed it down to the camera live latest version being the issue. On one mac (catalina) it was running fine (Camera Live v11), and on my laptop also (Catalina) it did not work (Camera Live v12). After downgrading to v11 on laptop everything is working properly.

Would be great if v13 solves this issue. Keep up the great work.

paulirish, 2020-09-08:
Another repro. Canon EOS M6 Mark II with a coupler & a/c adapter.
It only started once I upgraded to Catalina (10.15). It was fine on Mojave (10.14)

I saw "Err 70" on the display.

Also can confirm that rolling back to release 10 fixes the problem for me. (Didnt try 11)