v002/v002-Camera-Live #102: works for me 1 time but then doesn't let me select the camera on zoom/others

jlcanizales, 2020-04-12:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.4
 - Camera manufacturer and model: Canon 200D
 - Camera Live version: 13

**Describe the bug**
I use a few minutes ago and works great with cam twist and zoom, but then I open again Zoom but doesn't show any other camera than just FaceTime camera

**To Reproduce**
Steps to reproduce the behavior:
1. Connect camera
2. Turn on the camera
3. open camera live
4. select the canon 700d, to say active
5. open camtwist
6. select syphon
7. select camera live
8. open zoom 
9. go to camera settings and doesn't show (a few minutes ago show the other camera)

kmichaelshea, 2020-04-14:
What version of zoom are you using? 

I found the most recent update to zoom broke virtual cameras and I had to downgrade my zoom app for my 6D to work as a virtual camera again.

jlcanizales, 2020-04-15:
wow that remember me that the first time I test was with an old version of zoom (I don't know the version), then I decide to update to Versión:4.6.10 (20041.0408), the software due to some articles about stealing data and did work again, and as I read this I try with a new version (Versión:4.6.11 (20561.0413)) but also doesn't work

craigmdennis, 2020-04-15:
https://medium.com/@dkiesow/how-to-fix-the-virtual-camera-function-in-zoom-mac-only-9872a6d9ea5

theofanisamuel, 2020-04-23:
Mine also doesn't work on Zoom. Not showing any other camera rather than Facetime camera.
Currently I'm in macOS 10.15.4
Canon EOS 550D, CameraLive + CamTwist.

craigmdennis, 2020-04-23:
Install XCode. Remove the signature that enables hardware verification of cameras `codesign --remove-signature /Applications/zoom.us.app/`. Restart Zoom. Should work.

theofanisamuel, 2020-04-23:
> Install XCode. Remove the signature that enables hardware verification of cameras
> 
> ```
> codesign --remove-signature /Applications/zoom.us.app/
> ```
> 
> Restart Zoom. Should work.

Heyyy thanks man! It's working now, you're awesome!

joshspicer, 2020-05-02:
Just a note for anyone else searching around - removing the signature worked for me, and got my Nikon D3400 working with Zoom (along with CamTwist).  Picture was kinda poor, but functional!

joshspicer, 2020-05-02:
More interesting reads on entitlements and _why_ code signature needs to be removed: https://github.com/johnboiles/obs-mac-virtualcam/issues/4#issuecomment-609542482

mizu5588, 2020-05-18:
How did everyone get this installed. It only downloads a folder. I have tried downloading in several browsers. also tried running project in Xcode, but having all kind of issues there

bangnoise, 2020-05-18:
@mizu5588 this seems to be an occasional problem people have - see my answer in #135  https://github.com/v002/v002-Camera-Live/issues/135#issuecomment-630336917

natedigitalchumps, 2020-05-19:
Thank you for this! :)