v002/v002-Camera-Live #103: Canon 5D Mark III Low Frame rate lagging

lucamicheli, 2020-04-12:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version:10.15.3
 - Camera manufacturer and model Canon EOS 5D III
 - Camera Live version 13

**Describe the bug**
A clear and concise description of what the bug is.
I can see the video is lagging on my camera LCD and on the video on my mac

**To Reproduce**
Steps to reproduce the behaviour:
1. Just connect the camera and it will start lagging the video on LCD

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

scotcherontherocks, 2020-04-18:
Experiencing the same problem
macOS version:10.15.4
Camera manufacturer and model Canon EOS 5D III
Camera Live version 13

alew3, 2020-04-21:
Can confirm same issue.
MacOS 10.15.4
Canon 5dMkiii 
Camera Live Version 13

c6rbon, 2020-05-07:
Had exactly the same experience. Tested against a couple different siphon clients in case those were the issue.
Downgrading fixed it for me as well (thanks alew3!).