v002/v002-Camera-Live #104: Canon EOS Rebel T4i Battery Power is fine, External Power causes crash.

Daedalus686, 2020-04-13:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.4
 - Camera manufacturer and model: Canon EOS Rebel T4i
 - Camera Live version 11

**Describe the bug**
When starting the camera with battery, program usually works fine. When trying out an external power source this morning, program hangs and requires force quit, or crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Open program, then turn on camera with external power source installed
2. Program hangs indefinitely. Usually camera is shown in list, but will not select. OR, if you CAN select it, it displays "Camera Error"

**Expected behavior**
Turn on program, turn on camera, select camera, setup Camtwist.....

Daedalus686, 2020-04-14:
Tried a different USB port.... aaaaaand it now works. At least on first attempt. Will report again if it continues to have any issues.

Daedalus686, 2020-04-14:
And now it does not. Only worked the one time on a new USB port. Guess I should have left it all running!

hwschuur, 2020-04-14:
I experience the same with Camera Live version 11 on macOS 10.15.4. 
I expected the macOS 10.15.4 update to be the cause of this problem (it seemed to be the only change I made before the Camera Live stopped working; see issue https://github.com/v002/v002-Camera-Live/issues/85), but as a matter of fact there was another change: I bought an external power adapter to use my Canon 40D as a webcam.

Did some additional testing. Switched back to my battery and indeed: Camera Live 11 works fine again. Switched to external power supply and Camera Live 11 goes into 'Camera error' status.

Latest alpha build 13 doesn't seem to improve things.

Would be awesome if this could be fixed!

Daedalus686, 2020-04-14:
Alpha builds 12 and 13 both work.... kind of. They list the incorrect camera in the list. And the feed "lags", the framerate is very low.
Don't know if this helps at all. I'll continue to experiment, and in the meantime use the battery with build 11 until issue is eventually....hopefully resolved! :)

bangnoise, 2020-04-14:
It sounds like the external power supply, possibly in combination with the computer's power supply, is causing interference on the USB connection for both of you. The working but low-framerate output is presumably to do with tolerance within libgphoto2 - it is probably discarding some frames lost due to corruption - or pausing while they are resent.

There is nothing I can do in software to improve this for you apart from to improve the error reporting around it. Try different USB cables (they really do vary), try disconnecting other devices, avoid USB hubs, use high quality power supplies, avoid other causes of electrical interference.

Daedalus686, 2020-04-14:
Thanks for the advice, bangnoise! I'll try those things out today and report back. :)
Blessings

> It sounds like the external power supply, possibly in combination with the computer's power supply, is causing interference on the USB connection for both of you. The working but low-framerate [...]

hwschuur, 2020-04-15:
Thanks for the advice, @bangnoise! Any USB cables you'd recommend? In my case I'd need a USB-C cable to connect my Canon 40D and avoid USB hubs…

hwschuur, 2020-04-17:
Just to let you know, I bought a new USB-C to micro-USB cable to connect my camera and guess what: Camera Live (build 11) works like a charm again! 🎉

(Latest alpha build 13 works too, but the video feed seems to be a bit slower/more stuttering)

bangnoise, 2020-04-17:
@hwschuur great! What was your previous setup? Useful to be able to warn others if they have problems.

hwschuur, 2020-04-17:
Previously I connected indirectly via a Dell D6000 docking station.