v002/v002-Camera-Live #105: Camera Error on Sony A7R3

varunjani94, 2020-04-14:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.3
 - Camera manufacturer and model: Sony A7R3 
 - Camera Live version: 13

**Describe the bug**
All I can see is this screen:
[image]

**To Reproduce**
Steps to reproduce the behavior:

1. Connect the camera with USB C
2. Open Camera Live app and click on the camera that will show up i the window
3. The camera will click a photo at that time and show the error

**Expected behavior**
It should show active

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

obonbonbon, 2020-04-28:
Same error on Alpha7 ii

macOS version: 10.14.6
Camera manufacturer and model: Sony A7ii
Camera Live version: 13

markiespencer, 2020-05-13:
Same error with a Sony a7 (I) and MacOs 10.15.4.

Btw: Most methods seem to exclude this model since it lacks Live View, but I manage to get a functional live feed from the camera on my iPhone (via Sony's Imaging Edge Mobile app). I fail to see why the same would not be possible to Mac. Really hope you guys can get this working.

kalanga, 2020-09-09:
I am experiencing the same problem. The camera acts as though the shutter is being continuously depressed.

macOS version: 10.15.6
Camera: Sony A7 iii
Camera Live version: 13

bradsiefert, 2020-09-09:
I've got the same issue as @kalanga. Sony A7rIII.