v002/v002-Camera-Live #109: Apple iPhone 5 (PTP Mode)

mydudechris, 2020-04-17:
My iPhone X appears as a camera in the list of available cameras when connected via USB. It is showing as "iPhone 5 (PTP mode)

When trying to connect, CL throws a "Camera Error" message. Would it be possible to support iOS devices?

Here are the logs that appear when I attempt to connect to the iPhone

default	11:40:12.187427-0400	runningboardd	Acquiring assertion targeting executable<Camera Live(501)> from originator [daemon<com.apple.coreservices.launchservicesd>:143] with description <RBSAssertionDescriptor; frontmost:3068; ID: 326-143-1208; target: 3068> attributes = {
    <RBSDomainAttribute: 0x7ff391d33b00; domain: com.apple.launchservicesd; name: RoleUserInteractiveFocal; sourceEnvironment: 0x0>;
}
default	11:40:12.187587-0400	runningboardd	Assertion 326-143-1208 (target:executable<Camera Live(501)>) will be created as active
default	11:40:12.189445-0400	runningboardd	[executable<Camera Live(501)>:3068] Ignoring jetsam update because this process is not memory-managed
default	11:40:12.190589-0400	runningboardd	[executable<Camera Live(501)>:3068] Set darwin role to: UserInteractiveFocal
default	11:40:12.191421-0400	runningboardd	[executable<Camera Live(501)>:3068] Ignoring GPU update because this process is not GPU managed
default	11:40:12.192497-0400	runningboardd	Finished acquiring assertion 326-143-1208 (target:executable<Camera Live(501)>)
default	11:40:12.551237-0400	kernel	075238.430262 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	11:40:12.551683-0400	kernel	075238.430711 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	11:40:12.551963-0400	kernel	075238.430990 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	11:40:12.552515-0400	kernel	075238.431544 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	11:40:12.552995-0400	kernel	075238.432023 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
default	11:40:12.553267-0400	kernel	075238.432294 Camera Live@(null): AppleUSBHostUserClient::start: missing entitlement com.apple.ibridge.control
error	11:40:12.570166-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
error	11:40:12.570198-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
error	11:40:12.570218-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
error	11:40:12.570235-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
error	11:40:12.570251-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
error	11:40:12.570269-0400	Camera Live	gp_camera_capture_preview: -6 Unsupported operation
default	11:40:15.601552-0400	runningboardd	Invalidating assertion 326-143-1208 (target:executable<Camera Live(501)>) from originator 143
default	11:40:15.711143-0400	runningboardd	[executable<Camera Live(501)>:3068] Ignoring jetsam update because this process is not memory-managed
default	11:40:15.711195-0400	runningboardd	[executable<Camera Live(501)>:3068] Set darwin role to: UserInteractiveNonFocal
default	11:40:15.711256-0400	runningboardd	[executable<Camera Live(501)>:3068] Ignoring GPU update because this process is not GPU managed

theofanisamuel, 2020-05-09:
Mine also the same with iPhone 5. Is it possible to use it as a webcam also?
Any fix regarding this issue? @bangnoise

bangnoise, 2020-05-09:
This will be "fixed" in a future version - the iPhone won't show up in the list of sources. Google has existing solutions for using an iPhone as a webcam.