v002/v002-Camera-Live #110: Eos M50 not recognizes by Camera Live

Billigbrot, 2020-04-19:
**System:**
 - macOS version: 10.13.6
 - Camera manufacturer and model: Canon eos M50
 - Camera Live version 1.0.3

**Describe the bug**
I conact the camera to my macbook, with a usb. But camera live say: No camera. I had tested 4 usb cabels 

**To Reproduce**
Steps to reproduce the behavior:
1. Conatct the camera with the pc 
2. Open camera live
3. See no camera

bangnoise, 2020-04-19:
1.0.3 is not a Camera Live version. Try both the [alpha version 13](https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip) and the release [version 11](https://github.com/v002/v002-Camera-Live/releases/download/11/Camera.Live.zip)

libbythequill, 2020-04-19:
Just want to mention that Camera Live works well with my EOS M50 so it's not that particular model that's the problem - also OP have you messed around with the settings on your camera at all? I'm not positive but I think you should find the setting that says allow camera to be controlled over HDMI and turn that on. Even though you're using a usb (which is what I use too) I think that setting has to be enabled for Camera Live to work. Hope that helps!

Billigbrot, 2020-04-19:
Cool thanks, i think. The problem is the cable

aerogus, 2020-05-27:
1.0.3 is the version of the last firmware for EOS M50
My M50 is well recognized by Camera Live (big thanks for that) despite an error message every 30min cf. #150

bangnoise, 2020-05-27:
Closing this as a USB problem

closed 2020-05-27 by bangnoise