v002/v002-Camera-Live #114: Canon Powershot S90 "Camera error"

mateodelnorte, 2020-04-21:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.3 (19D76)
 - Camera manufacturer Canon Powershot S90
 - Camera Live version Version 11 (11)

**Describe the bug**

"Camera error" on connection.

**To Reproduce**
Connect camera, start app, select camera. 

**Expected behavior**
Not sure!

**Screenshots**

[image]

mateodelnorte, 2020-04-21:
Same error with version 13 Alpha:

[image]

I'm connected via USB.

jtragtenberg, 2020-06-04:
Same problem with Canon PowerShot G15