v002/v002-Camera-Live #115: Canon 80D stuck at starting

Todsapohn, 2020-04-23:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: [10.14.6]
 - Camera manufacturer and model [Canon EOS 80D]
 - Camera Live version [11]

**Describe the bug**
Camera Live shows status at "Starting" after I connect to camera and then it stuck this state.

**Screenshots**
[image]

canoneer, 2020-05-09:
This is the same as for my 60D - reported in later issue. Hope we can get some help.

bangnoise, 2020-05-27:
As an alternative maybe try https://www.usa.canon.com/internet/portal/us/home/support/self-help-center/eos-webcam-utility/