v002/v002-Camera-Live #116: Low Resolution at 320p

justryanliu, 2020-04-24:
Hi, 

i was wondering if there's a way to get a higher resolution at 1080p instead of 320p? I am using Camera Live with CamTwist and I could only get 320p resolution. Using the latest alpha 13 version.

Thanks!

isarmstrong, 2020-04-30:
I'm seeing the same result, looking through the settings to try to find a cause

isarmstrong, 2020-04-30:
@justryanliu

1. Change your resolution in CamTwist preferences
2. Shut down everything- your camera, CamTwist and Camera Live
3. Start your camera
4. Start Camera Live, wait for it to activate
5. Start CamTwist and add the syphon server

You should now be at the higher resolution.

bangnoise, 2020-04-30:
Thanks for the solution @isarmstrong. As this is to do with CamTwist, not Camera Live, I'm closing this issue.

closed 2020-04-30 by bangnoise