v002/v002-Camera-Live #118: Won't pick up Canon EOS Rebel T6 Camera

katsalom, 2020-04-29:
When I plug in my Canon EOS Rebel T6 nothing shows on Camera Live

**System:**
 - macOS version:  10.15.3
 - Camera manufacturer and model [ Canon EOS Rebel T6]
 - Camera Live version [11 & 13]

**Describe the bug**
When plugging in camera via USB I get no results in camera live. 

**Expected behavior**
I expected for my camera to pop up and be an option to click. 

**Additional context**
Cable picks up my g7x but it is not a supported camera at this time.

noahtallen, 2020-04-29:
I had the same issue, but after unplugging and plugging back in, it showed up in Camera Live but said "Camera Error"

lorenanicole, 2020-05-07:
I too am not getting my EOS Rebel T6 camera detected!

*System:*
* macOS Version 10.13.6
* Canon EOS Rebel T6
* Camera Live 11

*Bug:*
Connect to my MacBook Pro with the provided Canon cable (using usb to usbc adapters, as MacBook Pros only have the magical usb c ports), launch camera live and *no camera detected*.

*Expected:*
It to work 😄

lorenanicole, 2020-05-07:
Scratch that I found the solution! Hey folks per the [Canon forums](https://community.usa.canon.com/t5/EOS-Rebel/Rebel-T6i-won-t-connect-to-Mac-for-remote-shooting/td-p/250467), notice this beautiful GOTCHA:

>> Based on your description I recommend making sure the Wi-Fi feature of the camera is disabled. If the Wi-Fi is enabled the USB terminal of the camera will not output a signal to the computer. 

I disabled the wifi on the camera, and lo and behold it works!

[image]