v002/v002-Camera-Live #120: Camera Live Connected Canon EOS 40D not being recognized by CamTwist

Ender-54, 2020-04-30:
I am trying to get this up and running on my wife's new Macbook Air with Catalina running. I got the system to recognize the Canon EOS 40D fairly easily and downloaded the Alpha because of Catalina. While the camera is recognized, it is not showing up on CamTwist as a connected camera, even with Camera Live being listed as the Syphon server. After doing some research I saw that it appears Syphon has been disabled. My question is whether the camera not being visible as an option in the preferences page is due to Syphon being disabled (and at least for now being totally hosed), or if I am having another issue that I may be able to fix. 

Does anyone have this working on Catalina with CamTwist (using Syphon)?

bangnoise, 2020-04-30:
I'm not sure what you mean by "Syphon has been disabled". As this sounds like an issue with CamTwist, I'd suggest you contact them for support for their product.

You can check that Camera Live is working with any other Syphon app - eg it should show up in the [Simple Client](https://github.com/Syphon/Simple/releases/download/5/Syphon.Simple.Apps.zip) test app.

Ender-54, 2020-04-30:
My understanding is that Syphon doesn’t work anymore, since Mojave, because Apple removed the loophole to be able to use it. You are right though, it’s probably a CamTwist issue, I was just hoping it was something I did :(

closed 2020-04-30 by Ender-54

bangnoise, 2020-04-30:
Syphon still works (and uses documented Apple APIs, no loopholes). Good luck.