v002/v002-Camera-Live #121: Nikon D5500 not recognized (in any build)

hvsinc42, 2020-04-30:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.12.16 
 - Camera manufacturer and model: Nikon D5500
 - Camera Live version: Any/All

**Describe the bug**
I have plugged in my Nikon D5500 via USB (Nikon UC-E6), camera is not populating in Camera Live window it says "No Camera". I know the USB connection is working as I am able to see the camera in the Nikon Transfer 2 software and import functionality is working just fine. I tried this on my wifes laptop (MacBook Pro late 2018, OS 10.14.6) and its the same problem. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to finder, search "Camera Live"
2. Click on Camera Live
3. No Camera

**Expected behavior**
To be able to see and select my camera in the window.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

hvsinc42, 2020-04-30:
Apologies, looks like Alpha 13 did the trick, I said I tried all the builds which wasn't exactly true, I tried all the non-Alpha builds. Anyhow, looks like its working just fine now. Cheers!

bangnoise, 2020-04-30:
Great, yep, support for non-Canon cameras only appears in the recent alphas.

closed 2020-04-30 by bangnoise