v002/v002-Camera-Live #122: Aspect Ratio Issue

DanTheCameraMan, 2020-05-04:
- macOS version: 10.15.4
 - Camera manufacturer and model Canon 70d
 - Camera Live version 11

I use my canon 5d with this just fine but when I use a canon 70d at the same time by running duplicate version on camera live, the aspect ratio in camtwist has two black bars on either side of the image.

bangnoise, 2020-05-04:
Contact CamTwist for support for their product.

closed 2020-05-04 by bangnoise

DanTheCameraMan, 2020-05-04:
Thanks for your reply. Is it not then camera lives issue and its sending to camtwist

Thanks,
Dan.

DanTheCameraMan, 2020-05-04:
when I open syphon client app, it says im effectively getting a really small resolution out of my cameras feed of camera live. But there is no issue with my 5d iv or when I open Eos utility

bangnoise, 2020-05-04:
Can you describe the issue in the simplest possible setup using only Camera Live and Simple Client, and only describing what happens with the problem camera (not when it works with your other camera)? Please be precise wrt resolutions.

DanTheCameraMan, 2020-05-04:
When using a canon 70d connected through camera live, the simple client
says it’s giving a resolution of 960 x 640 which appears as a 3:2 aspect
ratio rather than 16:9. I have changed the resolution on the camera and in
1920x1080p 30fps it still appears as this box resolution in the simple
client.

bangnoise, 2020-05-04:
As per [Troubleshooting](https://github.com/v002/v002-Camera-Live#troubleshooting) in the README:

_The camera sends the image used for Live View - the dimensions will not match the camera's movie recording settings_