v002/v002-Camera-Live #124: Does it work with Nikon?

RealMrBread, 2020-05-06:
Objective = I want to use my camera as a webcam.
Camera    =  Nikon D5300
O.S.          =  macOS 10.14.6 

Found camera-live users making it work for a Nikon-Z6. I was wondering if it's on your roadmap to make it work for Nikon's D family.

Thanks!

bangnoise, 2020-05-06:
Support for Nikon cameras should be improved when I put out the next build.

meghdadFar, 2020-05-11:
Does it mean that the current releases do not support Nikon D5300? When can we expect to have support for this camera?

dcosson, 2020-05-21:
This is great, looking forward to the next release.

Can you share any info on what will be improved? I have a Z6 and it's working on the current version, but the quality is lower than I get from Ecamm Live (using the same usb feed from the camera in both cases). The resolution looks the same but it seems to have more aliasing or banding artifacts in Camera Live, especially noticeable when there's any movement.