v002/v002-Camera-Live #126: Not recognizing Canon EOS 7D

Jeff-Fitzsimmons, 2020-05-06:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.14.6 and 10.15.4
 - Camera manufacturer and model Canon EOS 7D
- Canon firmware 2.0.0 and 2.0.6 (latest)
 - Camera Live version 11 and 13

**Describe the bug**
I was using Camera Live with my Eos 7D for about a week. During that time I had the usual issues where I needed to turn it on and off or restart the program and turn everthing on in the exact right order, but once I got it going, it worked. 3 days ago, I went to use it and it is not recognizing my camera at all. I get a blank box with "No camera" in the header. 

I have tried rebooting, uninstall and reinstall camera live, unplug camera, turn on and off in multiple configurations (e.g. power on camera live first and then plug in, Plug in and power on and then open camera live, Power on and off while camera live is open, etc.) and I am getting no acknowledgement of the camera in camera live. I've tried different USB port and cable. Nothing at all.

I updated my OS to latest version 10.15.4 and updated camera firmware to latest and the behavior is the same - No response from Camera Live.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Camera live
2. Plug in Canon camera to USB port
3. Press "start/stop" to activate movie shooting
3. No response from Camera Live
(also tried reversing 1 &2)

**Expected behavior**
I expected Camera Live to recognize my camera and switch to 'Active' status

**Screenshots**
NA

Jeff-Fitzsimmons, 2020-05-06:
UPDATE: I ran through the entire troubleshooting path again and realized that I hadn't tried new cables after my OS and firmware update. When I replaced my USB extension cable with a new one, it worked.

closed 2020-05-06 by Jeff-Fitzsimmons

bangnoise, 2020-05-06:
Yep only one of my extension cables works - it's annoying. I'm hoping to improve error reporting around USB problems.

Jeff-Fitzsimmons, 2020-05-06:
I'm glad this happened. After updating OS and firmware (and replacing cable) Camera Live works every time.