v002/v002-Camera-Live #127: Fujifilm T2 / T3 & Sony A7 II

defcon1702, 2020-05-07:
MacBook Pro 15'' late 15
 - macOS Mojave: 10.14.6 
- Fujifilm X-T3, Fujifilm X-T2, Sony A7 II
 - Camera Live version 13
- connected directly with USB (no hubs)

The Fujis (both):
Work with USB Mode tethering auto -> but no Autofocus

Sony A7II -> when changing USB Type to "remote" (only mode that creates a connection) the cam shot an picture and creates the "camera error" error.

With all three cams together with Camtwist no connection in Jitsi or Zoom.

Streamlabs OBS is able to set the camtwist connection but without Autofocus.

andy-chin, 2020-05-15:
How to get that work? 
I plug directly to the Mac without connecting through a hub, no camera was detected.

USB Mode on USB Tether Shooting Auto

Rastafabisch, 2020-05-27:
@andy-chin this happened to me with a lower quality cable as well. once I chose a better one (admittedly hard to determine) I started working great. Which camera do you use?

samuelbriole, 2020-05-27:
@defcon1702 if you want to have continuous autofocus on your XT3, you need to enable the PRE-AF setting (source: https://youtu.be/wrU0TKKHLTM?t=74).

I have an issue with my XT3 though: the video is very laggy. My setup:
- late 2013 Macbook 15'' (max cpu and 16Go ram).
- Camera live 13
- Camtwist 
- OBS / Zoom

Do you know how to get a fluid video ? Maybe there's a way to lower the streaming resolution ?

andy-chin, 2020-05-28:
Ok guys, I got it working on my X-T2. And is working great. 

I am not sure what I did in between because after the first test which is not working, the next try is couple of days later, with multiple restart of the machine.

I used it in a Zoom session lasted for about 2 hours. I was worry about the heat and battery life, but I am surprised that the camera is not heated up and is almost chill :D Note that I am in a tropical country which is humid and summer all year long. And the battery also never drop any bit because I assume the constant charging thru the USB supply too.