v002/v002-Camera-Live #131: I installed Camera-Live on MacBook Pro and now Facetime, Zoom and Camera Live will not open.

FifeSing, 2020-05-09:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: [e.g. 10.15.3 - see Apple menu > About This Mac]
 - Camera manufacturer and model [e.g. Canon EOS 7D]
 - Camera Live version [e.g. 11]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

FifeSing, 2020-05-09:
Camera LIve v3 11
MacOS  10.11.6
I also installed CamTwist so thsi could be the problem.

Facetime crash details:

Process:               FaceTime [2051]
Path:                  /Applications/FaceTime.app/Contents/MacOS/FaceTime
Identifier:            com.apple.FaceTime
Version:               3.0 (2701)
Build Info:            FaceTime-2701000000000000~26
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           FaceTime [2051]
User ID:               501

Date/Time:             2020-05-09 13:21:15.831 +0100
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        1908CBE6-8C6D-4B08-BFF6-5DE5393AF789

Time Awake Since Boot: 4500 seconds

System Integrity Protection: enabled

Crashed Thread:        4  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0:
--> 
    __TEXT                 000000010662b000-000000010670b000 [  896K] r-x/rwx SM=COW  /Applications/FaceTime.app/Contents/MacOS/FaceTime

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff87ae96d2 __getattrlist + 10
1   com.apple.CoreServicesInternal	0x00007fff85976abf corePropertyProviderPrepareValues(__CFURL const*, __FileCache*, __CFString const* const*, void const**, long, void const*, __CFError**) + 488
2   com.apple.CoreServicesInternal	0x00007fff8596f6f7 prepareValuesForBitmap(__CFURL const*, __FileCache*, _FilePropertyBitmap*, __CFError**) + 227
3   com.apple.CoreServicesInternal	0x00007fff8596d357 _FSURLCopyResourcePropertyForKeyInternal(__CFURL const*, __CFString const*, void*, void*, __CFError**, unsigned char) + 212
4   com.apple.CoreFoundation      	0x00007fff9b7784fb CFURLCopyResourcePropertyForKey + 123
5   com.apple.CoreFoundation      	0x00007fff9b77c8e2 -[NSURL getResourceValue:forKey:error:] + 194
6   com.apple.Foundation          	0x00007fff8b68cc2d -[NSURL(NSURLPathUtilities) URLByAppendingPathComponent:] + 203
7   com.apple.AppKit              	0x00007fff8be49281 persistentStateDirectoryURLForBundleID + 288
8   com.apple.AppKit              	0x00007fff8be5127e +[NSPersistentUIManager(NSCarbonMethods) copyPersistentCarbonWindowDictionariesForBundleID:] + 195
9   com.apple.HIToolbox           	0x00007fff8cb01077 _HIPersistentUICopyPersistentCarbonWindowDictionariesForBundleID + 75
10  com.apple.HIToolbox           	0x00007fff8cb0101a GetPersistenceDictionariesAtLaunch() + 52
11  com.apple.HIToolbox           	0x00007fff8cb00fe0 WindowStateInitFirstEventTime() + 9
12  com.apple.HIToolbox           	0x00007fff8cb009dd _FirstEventTime + 52
13  com.apple.HIToolbox           	0x00007fff8cb0087b RunCurrentEventLoopInMode + 49
14  com.apple.HIToolbox           	0x00007fff8cb00677 ReceiveNextEventCommon + 184
15  com.apple.HIToolbox           	0x00007fff8cb005af _BlockUntilNextEventMatchingListInModeWithFilter + 71
16  com.apple.AppKit              	0x00007fff8be4fdf6 _DPSNextEvent + 1067
17  com.apple.AppKit              	0x00007fff8be4f226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
18  com.apple.AppKit              	0x00007fff8be43d80 -[NSApplication run] + 682
19  com.apple.AppKit              	0x00007fff8be0d368 NSApplicationMain + 1176
20  libdyld.dylib                 	0x00007fff973f35ad start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff87aea5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8b9c8578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8b9c6341 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff87aeaefa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff9a27a165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff9a279dcd _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff87aea5e2 __workq_kernreturn + 10
[... 1129 lines cut ...]
mapped file                       62.1M       18 
shared memory                     16.3M       11 
===========                     =======  ======= 
TOTAL                            584.4M      842 

Model: MacBookPro5,1, BootROM MBP51.007E.B06, 2 processors, Intel Core 2 Duo, 2.66 GHz, 8 GB, SMC 1.41f2
Graphics: NVIDIA GeForce 9400M, NVIDIA GeForce 9400M, PCI, 256 MB
Graphics: NVIDIA GeForce 9600M GT, NVIDIA GeForce 9600M GT, PCIe, 512 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x802C, 0x31364B54463531323634485A2D3147344D31
Memory Module: BANK 0/DIMM1, 4 GB, DDR3, 1333 MHz, 0x802C, 0x31364B54463531323634485A2D3147344D31
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 4.4.6f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 840 Series, 500.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-868
USB Device: USB 2.0 Bus
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus:

closed 2020-05-09 by FifeSing

bangnoise, 2020-05-09:
CamTwist is causing this crash. CamTwist requires macOS 10.12 or later.

Remove CamTwist.plugin from /Library/CoreMediaIO/Plug-Ins/DAL, remove com.allocinit.camtwistassistant.plist from /Library/LaunchDaemons, remove the CamTwist app and restart your machine.

If you're able to upgrade your OS version, it should work.

FifeSing, 2020-05-09:
Dear Tom:

Very many thanks for your reply - and a very quick one too. 
Your suggestion worked perfectly.

I suppose the advice given on the website that suggested the software should have specified the MacOS system required.
This was where I got the advice:
https://www.youtube.com/watch?v=CnOhBKpMkNk <https://www.youtube.com/watch?v=CnOhBKpMkNk>

It seemed very clear and well described. It says CamTwist v3.4.3 is suitable for MacOS 10.10 or higher. Presumably this is wrong.

Many thanks 

Peter Shepheard
Springthyme Records
Balmalcolm House, Balmalcolm
Cupar, Fife KY15 7TJ Scotland
tel: 01337 830773
www.springthyme.co.uk
progress@springthyme.co.uk
admin@springthyme.co.uk

bangnoise, 2020-05-09:
Yes CamTwist itself claims to support 10.10 and higher, however I've heard several reports of this exact problem on 10.11.