v002/v002-Camera-Live #135: Problem of installation

YARCHO, 2020-05-12:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.4 (mas os CATALINA)

Hello my name is Juan I am new to this platform (GitHub) 
I am trying to use my canon 6d as a webcam, that's why i got here. 
I download v002-Camera-Live but i don't know how to install it. Maybe you can help me.
Thank you very much

mizu5588, 2020-05-18:
same here. downloads as folder. even when downloaded on several different browsers

bangnoise, 2020-05-18:
Check the links you are following

Version 11 (stable): https://github.com/v002/v002-Camera-Live/releases/download/11/Camera.Live.zip

Version 13 (alpha): https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip

closed 2020-05-18 by bangnoise

bangnoise, 2020-05-18:
also I've had reports of Chrome failing to unpack the download, so definitely try another browser if you're using Chrome.