v002/v002-Camera-Live #136: Camera error in v13 - after previous success

msandman1943, 2020-05-13:
The first time I connected my camera to the computer, Camera Live recognized the camera, took a shot, and showed it as "active".  But since then, the app recognizes the camera, takes a shot, and then shows Camera Error.
Camera: Sony Alpha 7Riii with USB set to PC Remote
Computer: Macbook Pro, OS 10.15.4 (Catalina) 
See screen capture.
[image]
[image]

StefanoMassoli, 2020-05-13:
I have the same error with the Sony Alpha-A7S II
[image]

kdrachnik1, 2020-05-15:
i have the same issue with my Fuji X-T3.  I instantly get the Camera Error, it recognizes the Fuji, but once I click on Fuji i get the error.   Mac os 10.15.3

kimfar, 2020-05-16:
@kdrachnik1 For your X-T3, make sure you are not in a video mode and that your connect settings are USB tethered (Auto works well). I did a video on youtube explaining how to set this up and loads of comments and working through issues in the comments. Works well as a live view feed capture.

kdrachnik1, 2020-05-17:
Thanks I did watch your great video... my X-T3 shows USB icon on rear screen but still shows an error.  Do I need to download xcode to just get a live view w/ cameraOne?

KurtBeyers, 2020-05-18:
I've got also issues with my X-T3 camera, the setup went as described in the youtube movie by @kimfar .

However the camera is going through the following cycle in camera-live: active - camera error - active. Which causes the video to drop several times during a call in Teams or Zoom. 

My environment is the following:

MacOS 10.15.4
Fujifilm X-T3 firmware 3.2
Tether tools usb-c cable

Where can I find the log files to see where things go wrong when camera live looses the camera? Perhaps I'll learn more there. 

Thanks!

ariaarrazi, 2021-02-16:
i have the same issue with my sony a5100 , The first time I connected my camera to the computer, Camera Live recognized the camera showed it as "active". i can take a video streaming in video mode. But since then, the app recognizes the camera but alwas shows "Camera Error".

my environment:
MacOS 11.2 Big Sur
Sony a5100
USB cable (ANKER)
Camera Live 13
Cam Twist
OBS

georg2730, 2021-03-15:
I got the following problem:
My camera (Sony SLT-A58) ic connected to CameraLive and it says "ready". In Camtwist under Syphon it says: Syphon server and than only Camtwist is availible. 
When I open Zoom to try its not working, I can choose CameraTwist but the camera is black, but my cam is still on.