v002/v002-Camera-Live #139: Sony a6000 camera error

amitc0, 2020-05-15:
**System:**
 - macOS version: 10.15.4
 - Camera manufacturer and model: Sony a6000
 - Camera Live version: 12,13

**Describe the bug**
Camera live reporting camera error after ready state.
**To Reproduce**
Steps to reproduce the behavior:
1. Open Camera Live
2. Connect Sonya6000 camera  with USB cable
3. Wait till Camera Live Status recognise the camera and change the state to "Ready"
4. Double click camera name - State changing to "Connecting"
5. After less than a minute state change to "Camera Error"

pointmax, 2020-05-15:
I don't think sony cameras are supported. right now only canon cameras work

TayKangSheng, 2020-06-12:
fwiw. it only works for me using [12 alpha](https://github.com/v002/v002-Camera-Live/releases/tag/12) version

System: 
- macOS version: 10.15.4
- Camera model: Sony RX0

maxpointinger-brutkasten, 2020-06-12:
Oh wow, just tried Version 13 with my Sony A6500, works better than the Canon 600D (Rebel T3i).
The Sony A6000 doesn't seem to work with any version

jasongroulx, 2020-11-11:
Has anyone found a solution for the Sony A6000?