v002/v002-Camera-Live #140: Tools for overcoming audio delay?

jweath, 2020-05-18:
Hi there,

So grateful for this tool—thanks for all your work!

I'm using Camera Live + Cam Twist on a Mac to use my 7D Mark II with Zoom. My only issue with this setup is that audio delay (diff. between video and audio) seems to be quite bad—to my eye, more than the 120ms mentioned in the readme. Any settings / suggestions for decreasing latency, or other tools for adding an audio delay in a setup like this?

Thanks!

chrisspiegl, 2020-05-23:
You could setup OBS Studio to input the camera and the audio from your mic and there in the advanced audio preferences you can set a delay so that audio and video are in sync again. With that, you then can use OBS Virtualcam to use the OBS output as a Webcam source.

stackedsax, 2020-05-23:
Agreed with @chrisspiegl.  As an alternative if you use Audio Hijack, you could add a delay in AH and continue using CamTwist.  Audio Hijack isn't free, of course, so you may prefer the OBS route.

jweath, 2020-05-27:
Thanks! Also had some luck using Garageband with mic in, adding delay, outputting to Soundflower (which I selected as Zoom output)