v002/v002-Camera-Live #143: Low Quality Feed

ryanlaux, 2020-05-19:
Hi,

First off this utility is incredible, loaded up my Nikon Z7 and couldn't believe it was working. However I did notice the quality of the feed was quite low when loading the syphon server in camtwist. I'm on the latest v13 alpha since the previous versions don't seem to recognize my camera. 

I will note interestingly enough that I tried Ecamm for Mac (which recognized the z7 natively in much higher quality) but when I quit that app and re launched camera live, I was getting a much clearer image in camtwist. Not sure if it somehow is still defaulting to the Ecamm server, but if I quit camera live I do lose the image.

bangnoise, 2020-05-20:
Hi - could you repeat the experiment but without CamTwist? Use Simple Client ([download](https://github.com/Syphon/Simple/releases/download/5/Syphon.Simple.Apps.zip)) and see if you see a difference in resolution or quality before and after using Ecamm.

dcosson, 2020-05-21:
I was having the same problem with the Nikon Z6. I tried it out with this Simple Client, and it's showing 640x360 resolution, and hovering around 110-120 FPS.  On the Z6, it's set to use 1080p and 25 fps (though I'm not positive if it's actually sending at full 1080 to the USB out, those settings may only apply to internal recording?). But this may be part of the problem, that Camera Live is up-sampling the framerate so much? 

As far as the quality I'm seeing, 640 looks like lower resolution than what I'm getting in ECamm Live (I don't think ECamm will show me the actual resolution it's seeing). But aside from the resolution there is some kind of very noticeable aliasing or banding going on in the Camera Live video, especially when there's any movement. In ECamm Live that isn't happening at all.

bangnoise, 2020-05-21:
@dcosson Thanks for the info - the original reporter noticed that frame quality from Camera Live improved after quitting Ecamm - are you able to reproduce that effect? It would be good to know what you see in Simple Client if so.

ryanlaux, 2020-05-21:
Just did a quick screen cap so you can see what I'm seeing and the workaround. On first launch with camera live, simple client sees the lower res that dcosson reported. Once I open and quit eccam, then launch camera live again, resolution bumps up to 1024x576.

https://www.youtube.com/watch?v=gqtWHf9PnXc

bangnoise, 2020-05-21:
Thanks @ryanlaux - I'll see if I can work out a fix for the next release

ryanlaux, 2020-05-21:
@bangnoise Thanks appreciate your help! This utility has been a godsend

dcosson, 2020-05-22:
FWIW I'm seeing the same behavior, after opening and closing eCamm the resolution goes up, and the aliasing/banding thing stops. The frame rate is still at 110 FPS.

Thanks for looking at this, and I agree this app is super helpful!

felixjung, 2020-06-08:
Hi, Really helpful piece of software! Thank you. 👏 

I just want to add some more input to this issue. I'm successfully running Camera Live with my Fujifilm X-T2. The camera firmware is updated to the latest version (`4.32`) and I'm using the Camera Live 13 alpha.

The remaining problem I have is that the video grabbed from the camera is stuck at `640 x 480`. I've tried various different quality settings for stills in the camera:

* Image Size: `S`, `M`, `L`
* Aspect Ratio: `16:9`, `3:2`, `1:1`
* Image Quality: `FINE`, `NORMAL`

Unfortunately, none of these cause the output quality to be higher, but to affect the way the image gets letterboxed (i.e., black bars of different size at the top, bottom, left, and/or right side).

@kimfar, did you actually try Simple Syphon Client to confirm you are getting a 720p stream from your X-T3? As far as I can tell, software like CamTwist or OBS let you configure all kinds of things, but those settings don't change the resolution of the source stream. Would be a shame if this is really just a difference in support between the X-T2 and X-T3.

[image]
[image]

felixjung, 2020-06-08:
I have also tried Ecamm Live but it doesn't seem like there is a difference other than Ecamm Live not showing the letterbox.

kimfar, 2020-06-09:
@felixjung Yep on the XPRO2 I'm getting 1024X768 I no longer have an X-T2 to test out but would consider the only difference is the better resolution of the rear screen on the X-PRO2. The X-T3 is also getting 1024x768

felixjung, 2020-06-09:
> @felixjung Yep on the XPRO2 I'm getting 1024X768 I no longer have an X-T2 to test out but would consider the only difference is the better resolution of the rear screen on the X-PRO2. The X-T3 is also getting 1024x768

Interesting. Thanks for the feedback!

bangnoise, 2020-06-09:
Most cameras send the image they use for the live-view screen on the camera, so the dimensions from those cameras will be fixed to the screen's dimensions. Some cameras allow for some configuration of the image, which I need to improve support for so we get the largest available image.

BassBoomBox, 2021-05-01:
Hi guys, I am not sure if this is the place to ask and I am a computer noob, but I am just trying to make my Twitch stream look as good as possible with the means I currently have. I am using a Canon 550D (now called t2i) with Camera Live. I have a couple of questions:

I am loading this into OBS through Syphon (is this correct? I do not see another option, although I have tried adding CamTwist and then use video capturing device mode, without better results). 

I notice some frame rate loss. My camera is set to 1080p 30fps. I cannot change settings in Camera Live or OBS in Syphon mode. It's not a nice continuous view and I would like it to be. Please let me know if there is a middle man like the ones you're describing above. 

Thank you so much.