v002/v002-Camera-Live #145: Throttle frame-rate

bangnoise, 2020-05-21:
I was having the same problem with the Nikon Z6. I tried it out with this Simple Client, and it's showing 640x360 resolution, and hovering around 110-120 FPS...

_Originally posted by @dcosson in https://github.com/v002/v002-Camera-Live/issues/143#issuecomment-632343092_

scj643, 2020-06-22:
A side effect of the high polling rate causes high CPU using with libturbojpeg decompression. As shown here 
[image]

(From #165)

Using CFTimeInterval and having a check for passed time causes more CPU usage which isn't an option. Potentially NSTimers might be an option. One that fires on a set interval.

bangnoise, 2020-06-22:
@scj643 I have significant changes locally which will effect the solution to this, I'll push when I get a chance. Yes (obviously) don't poll, check target time for next frame after finishing a frame and set a timer if it's in the future.