v002/v002-Camera-Live #146: Occasional Memory Leak

ryanlaux, 2020-05-23:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.13.6 High Sierra
 - Nikon Z7 (was turned off at the time)
 - 13

**Describe the bug**
Computer locked up after leaving camera live on for a while in the background, checking ram usage showed camera live reporting 121 gigs of memory. Note z7 was off during this time, no feed was coming through. 

See attached file. 
[image]

bangnoise, 2020-05-25:
Oops.

Can you recall (or recreate) step by step instructions to arrive in this state? Eg did you turn the camera on and then off again? Had video been active prior to this?

ryanlaux, 2020-05-26:
Hmm can't recall any pattern to it but when this happened I had turned off my camera and left my computer alone/ locked, then came back and it had frozen up. Will monitor to see if any pattern emerges.

globeatin, 2021-01-28:
Any update on this issue? I have a Nikon Z6 on a MacBook running 10.15.7 Catalina. I am experiencing the same issue. Camera live showing 49GB of memory usage.