v002/v002-Camera-Live #147: App hanging in EdsOpenSession when building from source

dpkay, 2020-05-25:
Hi there,
I tried to build the app from scratch (thanks for amending the instructions!), so I can send a few EDSDK commands that are custom to my setup (exposure, white balance, use DOF, etc.) to the camera every time I start Live View.

Sadly, my build of the app seems to be hanging at "Starting..." in the UI, and in the EdsOpenSession call in startSessionOnQueue specifically. Things work fine with the release build. That's with a Rebel T2i if that matters.

Just curious if you've run into this? I tried both EDSDK 13.12 and 13.11, with the same result.
Thanks!

bangnoise, 2020-05-25:
Which OS version?

This sounds like an issue which has plagued the Canon SDK for some time - stability in the Canon SDK is the main reason I'm switching to libgphoto2 (see the libgphoto2 branch - you might be able to implement your changes there - afraid I haven't written instructions for building the required libraries yet).

If you pause in the debugger when it's hung you may be able to see the points of deadlock which might give a clue.

Sometimes having images on the camera's SD card has caused issues, try with an empty card fitted.

dpkay, 2020-05-25:
This is on Mojave 10.4.6. And great to know regarding libgphoto2, I'll give that a shot tomorrow, thanks! Let me know if there are any gotchas you remember off the top of your head (in the absence of documentation). And if that doesn't work, I'll dig a bit further to look for deadlocks.