v002/v002-Camera-Live #148: Does not work with Fujifilm X-T4

GregT1903, 2020-05-25:
**System:**
 - macOS version: 10.15.4 
 - Camera manufacturer and model Fujifilm X-T4
 - Camera Live version 13 V002

**Describe the bug**
Camera is not recognized get error
Tried every USB setting and camera is put to photo mode.

Seems to work for X-T3/X-T2 and X-H1, as X-T4 uses same processor I would believe there could be no problem. However the app gives error. 

Camera Error
USB PTP Class Camera.

Hope it can be fixed soon!

zwolfbites, 2020-05-25:
Having the same issue. Really hoping to use my X-T4 for a live stream soon. Thank you!!

Rastafabisch, 2020-05-27:
Working fine for the X-T2 on the latest firmware (4.32). Which settings USB tether mode do you use?

GregT1903, 2020-05-27:
Working fine on X-T2 and X-T3 NOT working fine on X-T4.
Auto as described, also tried the other modes. Just not working on the new X-T4.

davidhalldc, 2020-05-27:
Same issue with X-T4 as well.  I did see that today Fujifilm has released it's own app to turn GFX and X Series cameras into webcams (PC only).

GregT1903, 2020-05-27:
Yes PC only - unfortunately.