v002/v002-Camera-Live #150: canon m50 camera error issue

GazzaCazza, 2020-05-25:
after about 15-30 minutes my camera comes up with a screen saying to turn the camera off and on again or remove the battery. I have a canon m50 i bought less than a month ago and any help would be much appreciated.

aerogus, 2020-05-27:
I have the same (Canon EOS M50, last firmware 1.0.3) with a strange dirty error message (I will try to reproduce the bug and copy/paste the message here). I don't know if it's due to Camera Live or the device itself ). And it's also around 25/30min.

bangnoise, 2020-05-27:
Canon have released their [EOS Webcam Utility](https://www.usa.canon.com/internet/portal/us/home/support/self-help-center/eos-webcam-utility/) for Mac, which lists the M50 as supported - might work for you as an alternative.

GazzaCazza, 2020-05-28:
> Canon have released their [EOS Webcam Utility](https://www.usa.canon.com/internet/portal/us/home/support/self-help-center/eos-webcam-utility/) for Mac, which lists the M50 as supported - might work for you as an alternative.

the software seems much better and i havent run into the issue again using the webcam software, but still great job on the project guys, its really cool!

closed 2020-05-28 by GazzaCazza