v002/v002-Camera-Live #151: My Fuiji Xt20 keep getting Camera Error when starting.

BigBlilc, 2020-05-25:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.14.6
 - Camera manufacturer and model Fujifilm Xt20
 - Camera Live version 13

I have connected an Xt-20 and tried a GoPro both give me camera errors. Any setting I need to change?

Rastafabisch, 2020-05-27:
Try USB tether mode "auto" - if this doesn't fix your issue there is probably nothing you can do. It's especially worth noting that any camera working which not being included [here](https://github.com/v002/v002-Camera-Live/blob/master/CAMERAS.md) can be considered a nice side affect rather than working on purpose

mjigmond, 2020-05-29:
I can confirm this was the issue on a Fujifilm X-H1 (`auto` is not the default mode).

joshenders, 2020-06-10:
I can confirm that the Fujifilm XT-3 in USB Tether "auto" mode does indeed allow it to work as expected.

alexheuvink, 2020-07-25:
> I can confirm that the Fujifilm XT-3 in USB Tether "auto" mode does indeed allow it to work as expected.

Any chance you can tell me what other settings you are working with? I'm trying with my Fujifilm XT-4 and even in USB Tether auto mode, it's not working for me.
[image]

joshenders, 2020-07-26:
> > I can confirm that the Fujifilm XT-3 in USB Tether "auto" mode does indeed allow it to work as expected.
> 
> Any chance you can tell me what other settings you are working with? I'm trying with my Fujifilm XT-4 and even in USB Tether auto mode, it's not working for me.
> [image]

> Try USB tether mode "auto" - if this doesn't fix your issue there is probably nothing you can do. It's especially worth noting that any camera working which not being included here can be considered a nice side affect rather than working on purpose

MacOS Catalina, latest firmware of the XT-3 as if the timestamp. My guess is the XT-4 just happens to be unsupported. Manual focus actually didn’t work anyway. Not the best webcam.