v002/v002-Camera-Live #152: problem with canon m100

futurehomeless, 2020-05-26:
Hi, 

i install your software on catalina and it find my canon m100. I use cam twist than syphon than i choose cam live like the option. I try to use my camera like a source on obs program but i tried other program and the camera jerks and sometimes freezers. Your program write camera error also sometimes. How can i fix it?

jppellerin, 2020-05-27:
I think it's that the m100 is not a [supported camera](https://github.com/v002/v002-Camera-Live/blob/master/CAMERAS.md).  Mind you - I have that one and would love to see it supported!

rumncode, 2020-05-27:
M100 works fine for me using Alpha 13. Although I am getting a fairly low resolution (480 x 272 (16:9)). I don't have any freezing. Could be the USB cable?

futurehomeless, 2020-05-28:
> M100 works fine for me using Alpha 13. Although I am getting a fairly low resolution (480 x 272 (16:9)). I don't have any freezing. Could be the USB cable?

it's start to work than it give me that error but the camera go slow. i changed the resolution like you said but i have the same problem. I bought another cable today and tomorrow i figure out if it's a cable problem. Anyway im glad that m100 is working as a webcam because in the camera list don't appeare. I'll let you know when i receive the new cable.

thanks man

[image]

futurehomeless, 2020-05-29:
i change the cable and now it's work i just need to fix the black bars problem