v002/v002-Camera-Live #153: Fuji X-Pro3 not identified correctly.

alexjoellee, 2020-05-27:
- macOS 10.15.4
 - Camera : fuji x-pro 3.  v1.05
 - Camera Live version  13

when using a fuji x-pro3 in 'usb tethered auto' mode
camera live does not correct detect/initialize the camera
"
Camera Error
USB PTP Class Camera.
"
Mac then increases CPU use dramatically (fan spins up hard)

I would have expected the camera to identify as Fuji X-Pro3 and to enable connect as X-T2 does.

(appears to be the same issue as Fuji X-T4)

xprosarpedon, 2020-06-03:
Same problem here.

deryaerman, 2020-06-04:
I have the exact same problem although in my computer the ram usage increased dramatically to the point where mac asked me to close camera live.

vszalacha, 2020-06-06:
I can confirm the same thing.

alexheuvink, 2020-07-25:
Same problem with X-T4

phillipadsmith, 2020-09-01:
No problems here with the X-Pro 2.