v002/v002-Camera-Live #154: Canon M100 Low Resolution (Not CamTwist)

rumncode, 2020-05-27:
**System:**
 - macOS version: **Catalina 10.15.4**
 - Camera manufacturer and model: **Canon M100**
 - Camera Live version [e.g. 11] **13 Alpha**

**Describe the bug**
Low resolution of 480 x 272 (16:9), 480 x 320 (3:2) or 432 x 320 (4:3) reported directly in Syphon Simple Client depending on aspect ratio set on the camera. Same quality visually using Ecamm. Was hoping for a slightly higher resolution but I understand that it is limited by live view although the 3" LCD has a resolution of 1040k dots, the same as the Canon M50 from the spec sheets. 

I've tried 5 different USB cables and both video and photo mode. Disabled NFC & Bluetooth. Wifi not connected. 

Is it potentially a limitation of the software or can the camera just not output anything higher?

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on M100
2. Connect USB
2. Open Camera Live
3. Open Syphon Simple Client

**Expected behavior**
A higher resolution. Something more towards 720x480 - roughly translating dots to pixels.

**Screenshots**
[image]

**Additional info** 
No other issues with the software and it works very well (thank you - great project!)

scj643, 2020-06-11:
This is normal with libgphoto2, since it's only sending the preview over USB. Sadly the only way to get a higher resolution is to use an HDMI capture device if your camera can do a clean output.

rumncode, 2020-06-12:
Yep - think it must be the limitations of the preview via USB. Useful for others with the M100 - I ended up getting a capture card which is running 1080p 60fps now from the camera. Using a clean HDMI script from the CHDK forum on the memory card which also removes the 30 minute camera shut down.