v002/v002-Camera-Live #155: Camera Error

chrisb1081, 2020-05-28:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: [Sierra 10.12.6]
 - Camera manufacturer and model [Canon EOS M10]
 - Camera Live version [13]

**Describe the bug**
Camera Error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

themufi, 2020-06-02:
I have exactly the same error. Only, I'm trying to use an EOS rebel t1i ( also known in some countries as the 500D)
[image]

bangnoise, 2020-06-02:
@chrisb1081 I'm not sure the M10 works, I'll try to report that better in the next version
@themufi try everything in the [Troubleshooting](https://github.com/v002/v002-Camera-Live#troubleshooting) section of the README.

themufi, 2020-06-03:
Hey Tom,

No luck with the troubleshooting guide either - tried everything.

The alpha version of cam live picked up my device as EOS 500D

BluesMessiah, 2020-08-24:
Well, thought I'd try and see if the Cannon Powershot  SX530 HS would be supported.  It picked it up correctly but immediately gave me the Camera error message, so I'm guessing I am out of luck. (yes, I know it isn't listed in the supported cameras).