v002/v002-Camera-Live #158: Recognizes 7D Mark II But Can't Use in Zoom

jasongurwin, 2020-06-05:
Camera Live activates for my 7D Mark II, but when I go to Zoom it doesn't show up as an option.

[image]
[image]

vadalus911, 2020-06-08:
read the instructions. Use CamTwist or OBS with a virtual driver to get it into Zoom etc..

redraushan, 2020-06-26:
@vadalus911 same thing is happening with my canon 5d mark iii, I tried to install CamTwist and selected Camera Live from the list still I can't find in the list of cameras in Zoom and Skype. 

However, for web based applications like skype and google meet, CamTwist is available in camera list and working perfectly. There seems to be some issue with installed applications only.

Machine: macOs Catalina 10.15.5