v002/v002-Camera-Live #159: Can't download v002-Camera-Live to Mac

davidacole, 2020-06-05:
I've now used three different browsers and it downloads either as a zip or a folder. When those are opened there is nothing there that opens the app

bangnoise, 2020-06-05:
What link are you downloading?

davidacole, 2020-06-05:
https://github.com/v002/v002-Camera-Live.git <https://github.com/v002/v002-Camera-Live.git> I think…

I hit the green button here…

What e
David A. Cole
214.728.3817

bangnoise, 2020-06-05:
Ah no, that's a link to the source via git.

Version 11 (stable, only supports Canon cameras, has some issues)
https://github.com/v002/v002-Camera-Live/releases/download/11/Camera.Live.zip

Version 13 (alpha, supports more cameras, has different issues)
https://github.com/v002/v002-Camera-Live/releases/download/13/Camera.Live.zip

closed 2020-06-05 by bangnoise