v002/v002-Camera-Live #160: Is there any documentation around what resolution to expect from the camera?

vadalus911, 2020-06-08:
It does not seem to have anything to do with the settings in the camera itself?

Canon 1300D. Can set the video resolution to whatever you want but the Syphon Simple Client will always show 1008x672@30fps.

scj643, 2020-06-11:
The camera preview is usually a low resolution. You can get the preview resolution by doing the following in terminal with gphoto and mpv. `gphoto2 --capture-movie --stdout  | mpv --demuxer-lavf-format=mjpeg -` Press I to show the resolution.

vadalus911, 2020-06-11:
Yep thanks. The simple syphon client also shows the resolution which I assume is native. Agree it’s relatively low... but ok.. around 700p

scj643, 2020-06-11:
Still cheaper than buying a proper HDMI capture device right now. People are scalping the Elgato Cam Link 4k for about $50 over retail. Also you do get one extra benifit doing it this way. You can control the camera settings over USB once that is implemented. There is no way (at least on my d5600) to have live view over HDMI and USB control.

sumade123, 2020-07-09:
I have a Panasonic FZ1000. I see that this software supports only Canon cameras (although some people seem to be mentioning other brands in the issues, can someone clarify if others work?)

I guess my question is, given that the FZ1000 seems to be a reasonable camera, would it be better for me to buy an Elgato Cam Link or could I expect better results buying a cheap used Canon camera from the list on Camera.md and using Camera Live?

Thanks

scj643, 2020-07-10:
@sumade123 Looking at gphoto2's camera support it looks like it doesn't support the Panasonic FZ1000 line for live view so it wouldn't work with the alpha build. 

Also using a Elgato Cam Link will always provide better results, since most cameras limit the resolution over USB. For my use case though, using a cam link is actually not feasable, since I have added camera controls to Camera Live so I can remotely control my camera. At least with Nikon it is not possible to get a clean HDMI output and be connected to USB.