v002/v002-Camera-Live #164: Cropped View (NOT full in ZOOM App)

Dr3wskjie, 2020-06-18:
Hi Guys,

I have been using the following setup since this whole pandemic started in the country I'm currently living for ZOOM OnLine Meetings:
Softwears:
- CamLive
- CamTwist
- ZOOM
Hardwares:
- Canon 7D Mark I

Everything is still working perfectly fine. My only issue left IS:
- In Zoom, my frame is cropped on the sides (left and right)
- I've gone to video settings in Zoom App and changed "Original Ratio" to "Widescreen" (16:9) and it still appears the same. 

I'm thinking, this could be the settings either on CamTwist or CamLink. I have searched on each app to see/find some kine of settings but I have had zero luck. 

I'm crossing my fingers and really hoping that someone can help or share any ideas or solution to this issue that I've been trying to figure out for the past month. 

Thank You so much in advance.

bangnoise, 2020-06-18:
This is an issue with CamTwist. See the solution in #44

closed 2020-06-18 by bangnoise

Dr3wskjie, 2020-06-19:
Thank you for the quick reply. 

I have successfully gotten the issue resolved by following the instruction using Simple Live to detect the resolution that my camera is outputting. Thank you so very much for this. 

However, my question about getting the camera to be autofocus instead of adjusting it manually has not yet been answered. 

Anyone with this experience (Re: autofocus) using CamLive and CamTwist, please kindly share here. 

Thanks again guys.... :D

bangnoise, 2020-06-19:
No support for autofocus atm - it's on the list of wanted improvements.

Dr3wskjie, 2020-06-20:
Ahh... I see... I hope it's in the works, because that would really help
out in a very GREAT DEAL. Not sure if anyone else had address this request
recently.

Thanks again for the prompt response.

*Drew*

Dr3wskjie, 2021-08-30:
Hi Tom,

How have you been ? Hope all is well.

I was just wondering if there is/are any updates that I'm not aware of
regarding Camera Live. Reason why I'm asking is because I have been using
the same one ever since we started communicating, I have been so afraid to
do any "Updates" for Zoom.

I look forward to hearing back from you.

Thank you in advance.

Warm Regards,
*Drew*