v002/v002-Camera-Live #166: Least expensive supported camera?

rdlauer, 2020-06-26:
The list of supported cameras is fantastic, but for noobs like me it would be helpful to also know which cameras are generally the least expensive. Anyone happen to know which of the supported Canon models are generally the least expensive on ebay etc?

ryanlaux, 2020-06-29:
This is pretty subjective.. but you may have luck searching for older canon bodies such as the T3i or T4i. older. you get the least quality you'll get of course. Also comes down to lens choice, for decent low-light performance, and a nice depth of field (I.e. blurry background) you'll want a lens that has an aperture of f1.8. a 35mm or 50mm would do nicely, just depends on how wide you want your shot to be.

rdlauer, 2020-06-29:
Thanks, that's helpful!

closed 2020-06-29 by rdlauer