v002/v002-Camera-Live #167: Camera freezes periodically (Nikon D5100)

allenjarvis, 2020-06-30:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 10.15.5
 - Camera manufacturer and model: Nikon D5100
 - Camera Live version: 13 (Alpha) -- Build 11 does not show my camera as an option at all

**Describe the bug**
While I am able to start my camera and see the output in Cam Twist, Camera Live seems to hang up roughly each hour. I have to quit Camera Live and restart it to unfreeze the video.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Camera Live
2. Select Camera
3. Start streaming video for Google Hangouts
4. Have my face freeze in at a ridiculous-looking moment
5. Quit and restart Camera live to resolve

**Expected behavior**
Not freezing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

bubba-ltn, 2020-07-18:
I am having the same issue with Canon M50s.

gokeji, 2020-08-17:
Same issue with Nikon D610, freezing more often than once per hour.

dsmith001, 2020-09-01:
Same issue using Camera Live along with CamTwist and for streaming video on Zoom. Restarting Camera Live solves the issue.  This *seems* to only happen while streaming through Zoom (haven't used any other streaming/video platform).  Will happen sometimes 4-5 times within a 30 minute period. While not streaming Camera Live seems to be fine (as evidenced by the preview window on CamTwist continuing to show a live picture.

tanmaywadhwa, 2020-09-17:
+1. Seeing a similar issue on Nikon D3400

antonkeller, 2020-10-21:
Same here with canon M50, the span before crash is between 10 and 25 minutes.

aranpura, 2020-11-15:
+1 I see this issue about once per hour with Camera Live v13 (alpha) and a Nikon D610. If I just quit and re-launch Camera Live everything is unfrozen.

youhan89, 2021-04-20:
+1 Seeing this on my Nikon D5100 aswell. Works fine for about 15-30 minutes then video freezes. As many others i use Camera Live in combination with CamTwist mainly used in Microsoft Teams in browser. Restarting both applications fixes the issue for another 15-30 minutes.