v002/v002-Camera-Live #173: 2-3 seconds lag every 15 seconds on Mojave

jrucho, 2020-08-19:
- macOS version: 10.14.6
 - Camera manufacturer and model : Canon 7D, 5DmkII and 550d
 - Camera Live version : 11 and 13

Hi! I used Camera Live for a few years now and I was mainly using it on my old macbook pro from 2012 on Sierra and everything was working as expected ( even with two cameras at the same time with two separate instances of the app). I just found out after upgrading that macbook pro and another macbook 12" to Mojave that the app (v11) doesn't work the same anymore. It starts and after e few seconds there is two or three seconds of lag happening every  now and then. It is like a patern (10-15 seconds fluid and then a few seconds of lag). It happens with my three cameras.

Steps to reproduce the behavior:
1. Connect the camera.
2. Open the app.
3. Select The camera and wait for it to start with the shutter sound.
4. Open Syphon Recorder
5. See the result.

I expected to get a clean feed like on Sierra but some of the video codecs might have changed since then. 
Do you know if there is a memory problem on Mojave? Does it work better in Catalina?

I used the v13 of the app and I found out that the app is more responsive and fast but the lag it higher than the v.11 for me, for the 3 cameras and the two macbooks. 

ps: On Sierra and High Sierra , I could get a lag free feed on my macbook pro 2012 for hours and 15 min of lag free on my 12" macbook while recording at 720p inside Syphon Recorder.

Cheers

snicolai-blog, 2020-08-30:
Something in Mojave changed in the past several months and I'm seeing this now as well, I didn't see it in April/May in an older Mojave.

I suspect that the default GCD priority changed (or something else around it changed priority, causing the default to get less time).  I see the queue is created with default priority, not user interactive, which might help alleviate this problem.  See 

https://github.com/v002/v002-Camera-Live/blob/master/Camera%20Live/SyPAppDelegate.m#L136

It's an easy fix to use

https://developer.apple.com/documentation/dispatch/1453028-dispatch_queue_attr_make_with_qo

to change the QoS to user interactive.

jrucho, 2020-09-02:
> Something in Mojave changed in the past several months and I'm seeing this now as well, I didn't see it in April/May in an older Mojave.
>
> I suspect that the default GCD priority changed (or [...]

Does this need to be changed and compiled in a new app or could it be possible to activate user interactive on this particular app using Terminal?

jrucho, 2020-09-08:
I am doing some testings right now. I think my problem might be caused by the HFS+ to AFPS change (from Sierra to Mojave). I did a clean install of the OS and I still get some lag. Tested with fresh Mojave and High Sierra installs (AFPS) and various USB cables that were working perfectly on Sierra. I am going to install Sierra back, next. I ll keep you updated.