v002/v002-Camera-Live #175: Crashes during livestream

filipetelles, 2020-08-31:
Sometimes the program crashes and it is only possible to open it again after restarting the system.
I use it mainly to make live streams. Then the error causes the transmission to end.

I would like to know if there is any task that I could end instead of restarting the system.

**System:**
 - macOS Catalina 10.15.5
 - Canon EOS 5D Mark III and Canon EOS 6D
 - Camera Live version 11

**To Reproduce**
Steps to reproduce the behavior:
Every time the camera disconnects accidentally or for lack of battery and rarely for no reason.

**Expected behavior**
Reopen the program without having to restart the system.

apbarratt, 2020-10-09:
Same here, but crashing seems to be fairly consistent, usually after the camera has lost connection due to something like battery loss.  I've a wedding tomorrow and can't really afford to be restarting my computer every time this happens!

PaulHobbs, 2020-10-14:
Same, Camera Live crashes whenever I turn off the camera or it loses power, and the only way to fix it is to restart my computer.