v002/v002-Camera-Live #178: Recompile with new version of canon SDK for extended camera support

marsson, 2020-10-26:
New cameras on the current build are not supported (probably due to the lack of support on the compiled version of the canon SDK)

If any of the owners of the repo have access to the current new version (13.13.0) and would recompile the app with the new version, it would potentially add support to the new models (R5, R6, etc) 

I was going to fork the project and add myself, but here in Australia we live in the past and have no access to the new SDK. (thank you canon).

Alternatively, if anyone would provide me with the current SDK, I would compile it and test/fix issues myself and pull request it back to the repo. (really, I would be glad to help).

kurtisdunn, 2020-11-05:
Canon R6 not working. An update would be appreciated to no end!

meetbryce, 2023-01-12:
@marsson where would I get the SDK? happy to get a copy for you