v002/v002-Camera-Live #179: Big Sur issues - slow to open and high cpu than in Catalina

mangoman108, 2020-11-13:
Ive just updated to Big Sur.

Camera Live is working but I have noticed performance issues. Connecting with a Canon 5d Mkiii - been using CL for months (and loving it!). Since upgrade it is slow to load and CPU 50%+

Just letting you know - am sure you are on it or will be soon enough if others are experiencing same.

I'm on MBP 15in 2018 and running CL 11.

Thanks for your efforts in making such a great and useful app.

antoniojl, 2021-01-29:
[image]

A little example. Running a Canon 7D into Camera Live on my 2020 MacBook Pro.

antoniojl, 2021-02-09:
Version 13 Alpha fixes this for me: https://github.com/v002/v002-Camera-Live/releases/tag/13