v002/v002-Camera-Live #180: M1 compatibility?

alialkhatib, 2020-11-22:
Just curious if Camera Live works for people with computers running M1 silicon. Apple has been emphatic about Rosetta 2's compatibility, but I'm reading that the areas that are causing issues tend to be with peripherals and whatnot, and I think this is the most important piece of software that I use (of the ones I'm worried about).

Ideally it'd be great to hear that there's an ARM version of Camera Live coming at some point. It'd also be fine just to hear that Camera Live works through Rosetta or something.

In the meantime, I'm holding off on buying a Mac Mini, but in the long term I guess I could keep a legacy device around to run this. I'd just rather not keep a device around for that one purpose.

I'm specifically looking at [this article](https://www.forbes.com/sites/patrickmoorhead/2020/11/21/apple-macbook-pro-13-m1-reviewwhy-you-might-want-to-pass/) and [this site detailing which apps are compatible, working through rosetta, or not working at all](https://isapplesiliconready.com/). i didn't see Camera Live listed, so I figured I'd just ask.

bangnoise, 2020-11-22:
I'll be able to test mid-December, or perhaps someone else will chime in before then.

YuriyNasretdinov, 2020-11-27:
I can confirm that both Camera Live 11 and Camera Live 13 work through Rosetta although not without issues. E.g. Camera Live 11 gives stable ~30 FPS on my M1 MacBook Air but it also consumes a whole 1 CPU core which may or may not be a problem for you. Camera Live 13 does work as well but gives around ~15 FPS but consumes much less CPU.

hsdcreative, 2021-02-01:
Hey, any updates on the Mac m1 update. I really love your product and rely on it to connect my DSLR.

alialkhatib, 2021-02-01:
as it happens i got an M1 Mac Mini (base configuration) in the last few days and it works without any issues that i can observe. this is with a Canon 6D, if that makes any difference

hsdcreative, 2021-02-01:
Oh awesome thanks for the feedback. yeah the app seem to open and work fine. but it is not realizing my Nikon. I have both D3300 and D750, only tested the D3300 and it works fine on my MBP that runs Catalina. my M1 Mac mini (also base configuration) isn't able to read it. but I was able to work my way with installing OBS + CamTwist + Nikon Webcam utility and use webcam instead of syphon in CamTwist to bring up the image form the Nikon webcam utility, resize it>> send it to OBS and then OBS start virtual camera>> select OBS in zoom. 

That is the only way it worked for me using the USB cable. The zoom somehow wasn't able to recognize CamTwist like it did on my MBP (Catalina OS). the picture quality still doesn't seem to translate well using that method on M1, while it did on the MBP when (I adjusted settings in CamTwists to match what the camera is set to in both M1 mini and the MBP). 

Side note, although I didn't see anything about Nikon when reading the compatibility for live Cam, it seemed to recognize my Nikons and iPad/ iPhone but not my Sony a600! this is interesting. maybe I'll finally cave in for a video HDMI capture card!

Jentzell, 2022-10-10:
Please correct me if I'm wrong but it looks like it only works in rosetta emulation. M1 native apps (for example Teams) can't find the webcam driver. So in other words there needs to be added arm64 native binaries as well to get full support for the M1.

YuriyNasretdinov, 2022-10-10:
That is correct, it works in both directions I think, if you only have an
Apple Silicon native driver for the webcam, and no Intel version, it won't
work in Rosetta apps. The Intel and the Apple Silicon versions work as two
separate drivers essentially

xiaokunx, 2023-01-27:
Have the same problem. On the old intel cpu Mac, the Canon T1i (500D) -> Camera Live -> CamTwist -> Zoom/GoogleMeet works. But on the new M1 cpu Mac, even though the Camera Live feed could be recognized by CamTwist via siphon source, the CamTwist virtual camera is not recognized by any of the meeting applications. 

Turned out if you check the driver from CamTwist (3.4.3, from https://camtwiststudio.com/download/), it is compiled for intel chips (x86_64). 
```
> file /Library/CoreMediaIO/Plug-Ins/DAL/CamTwist.plugin/Contents/MacOS/CamTwist
/Library/CoreMediaIO/Plug-Ins/DAL/CamTwist.plugin/Contents/MacOS/CamTwist: Mach-O 64-bit bundle x86_64
```

After much struggle and search, I found OBS (https://obsproject.com/download) might be the alternative. Notice it does have an apple silicon installer. and it finally worked in a similar way Canon T1i (500D) -> Camera Live -> OBS -> Zoom, but unfortunately still doesn't work for GoogleMeet run in the browser.
```
>file /Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam
/Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam: Mach-O universal binary with 3 architectures: [x86_64:Mach-O 64-bit bundle x86_64
- Mach-O 64-bit bundle x86_64] [arm64:Mach-O 64-bit bundle arm64
- Mach-O 64-bit bundle arm64] [arm64e:Mach-O 64-bit bundle arm64e
- Mach-O 64-bit bundle arm64e]
/Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam (for architecture x86_64):	Mach-O 64-bit bundle x86_64
/Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam (for architecture arm64):	Mach-O 64-bit bundle arm64
/Library/CoreMediaIO/Plug-Ins/DAL/obs-mac-virtualcam.plugin/Contents/MacOS/obs-mac-virtualcam (for architecture arm64e):	Mach-O 64-bit bundle arm64e
```

In the OSB UI's `Source` panel, click `+` and choose `Syphon Client`add the `[Camera Live] Canon EOS 500D` as the Source. Then in the `Controls` panel, configure the `Start Virtual Camera`, chose the `Output Type` as `Source`, and for `Output Selection` accept the Source Name you just picked.