v002/v002-Camera-Live #181: Camera Live Camera not available under Big Sur

bmdev88, 2020-12-16:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: 11.0.1 (20B29)
 - Camera manufacturer and model Canon EOS M50
 - Camera Live version: Version 13 (13)

**Describe the bug**
After starting Camera Live i can see my Camera and also that it is active - but i've no glue why i can't choose the Camera Live Camera within OBS / CamTwist / MS Teams....

**To Reproduce**
Steps to reproduce the behavior:
Start Camera Live - everything fine but no Camera within Streaming / Videoconferencing solution

**Expected behavior**
Use Camera Live as a Video Source in OBS

**Screenshots**
[image]

Thanks!

NickyMateev, 2021-01-19:
Hi @bmdev88, I've got the same setup as you (camera + osx), in CamTwist I can select the camera, also in Zoom, but in Teams I cannot..

pablonoel, 2021-05-14:
Interentingly I didn't had any problem until yesterday, when I updated to Big Sur 11.3.1. 
Camera live does recognize the camera properly, but the syphon server is not available anymore to CamTwist.

I did reinstall the latest (alpha 13) Camera Live and the latest CamTwist with no avail.

MacBook Pro (15-inch, 2017) - 3,1 GHz Quad-Core Intel Core i7
Canon 6D with tether pro cable usb-mini to usb-c

rickardholmberg, 2021-08-03:
TLDR: Works for me in 11.5.1.

I had been putting off my upgrade to Big Sur exactly because of uncertainties around Camera Live and CamTwist issues. Today I took the leap, and to my surprise my setup with an old Canon EOS 550D, Camera Live, and CamTwist worked without any changes whatsoever! So far I've tested Microsoft Teams (works), Zoom (works), Signal (works) and Facetime (didn't work in Catalina, still doesn't see CamTwist).

I have built Camera Live myself, though, from the branch https://github.com/rickardraysearch/v002-Camera-Live/tree/rickardraysearch/isospeedmenu , but that really shouldn't make a difference.

MacBook Pro (13-inch, 2016, Four Thunderbold 3 Ports) - 3,1 GHz Dual-Core Intel Core i5
Canon EOS 550D with USB tether to an old Acer USB-C Dock (where the network interface has been a dud since 10.11...)