v002/v002-Camera-Live #189: Issues on OS X 14.1 Sonoma? Stopped working after my upgrade

jestelle, 2023-11-02:
**PLEASE COMPLETE ALL OF THIS SECTION**

**System:**
 - macOS version: [e.g. 10.15.3 - see Apple menu > About This Mac]
14.1
 - Camera manufacturer and model [e.g. Canon EOS 7D]
 Nikon D5100
 - Camera Live version [e.g. 11]
 13

**Describe the bug**
I use Camera Live with Camtwist, and since I upgrade to Sonoma I don't see any cameras available where previously I saw my Camtwist camera.

Camtwist does still seem aware of Camera Live...

tcmendenhall618, 2023-11-28:
I just ran into the same issue. After doing some digging, it appears that Apple has change some secure camera settings that may be impacting older devices without modern extensions. Here are the instructions from Apples website (https://support.apple.com/en-us/HT213969) that I used to get Camera Live to see the camera again: 

Pasting the instructions as well in case the link breaks in 6 months.
-----
Restore legacy video device support
When you restore these legacy settings, the camera privacy indicator won't display in the menu bar while a camera with older software is in use.
1. Restart your Mac into Recovery Mode:
2. Click the Options button, then click Continue.
3. From the Utilities menu, choose Terminal.
4. Enter this command:
system-override legacy-camera-plugins-without-sw-camera-indication=on

5. Press Return. Depending on your system configuration, you might be prompted to complete additional steps.
6. Quit Terminal.
7. From the Apple menu, choose Restart.

To change back to the default settings, follow the steps again and enter this command:
system-override legacy-camera-plugins-without-sw-camera-indication=off

Check the Privacy & Security settings for Camera
If the legacy video device support has been restored on your Mac, a warning appears in System Settings:
1. Choose Apple menu  > System Settings.
2. Scroll down, then click Privacy & Security.
3. Click Camera. If legacy video support has been restored, you get a warning that the camera privacy indicator won't appear in the menu bar when a camera or video output device with an unsupported plugin is in use.

acramatte, 2023-11-29:
I have a similar setup with Camera Live + CamTwist that stopped working after having updated to Sonoma 14.1 for the reasons @tcmendenhall618 mentioned above.

As you mentioned @jestelle, the Syphon server from Camera Live still works normally so the problem is most likely coming from CamTwist that hasn't received any updates in a while.

I have now replaced CamTwist by OBS Studio, with a _Syphon Client_ using Camera Live as a Source:
[image]

There might be alternatives to OBS, but it's just to say the issue is not directly from Camera Live.

bangnoise, 2023-12-13:
Yep, Camera Live has nothing to do with the virtual camera provided by CamTwist. This would be a CamTwist issue.

I'm sure this will be useful info for others though, thanks for the comments everyone.

closed 2023-12-13 by bangnoise